FRC2706/2020-2706-Robot-Code #10: USB Camera

nicojopr, 2020-01-11:


kevlam2706, 2020-01-15:
We can directly support a USB camera plugged into the RoboRio and generate a CameraStream that is sent back to the driver station laptop. It can be viewed in a dashboard such as Shuffleboard.

However, the Vision team also supports streaming camera signals, coming from the Raspberry Pi that is driving the external vision system. This can also be streamed to the driver station laptop, and viewed in something like a web browser or possibly also by a dashboard tool such as Shuffleboard.

Why choose one approach over the other? We need to consider:
- reliability
- latency
- frame rate
- field of view (widescreen vs square image views)
- bandwidth, which will be limited by FMS to about 4 mbps total
- image size (from 1920x1080 all the way down to 160x120 - what's the best we can get away with?)

Whichever approach(es) we support, we need to provide the code, if any, to initialize it, and set it up on the driver station.

EdenZ-2706, 2020-02-08:
I tried to run the USB camera on the RoboRIO, and tried with different resolutions. 
Ethan T. also did similar things in the preseason.
With higher resolutions, there was latency and bandwidth was an issue. 
To get within reasonable bandwidth/latency, resolution had to be pretty low.
A pi would be ideal.

closed 2020-02-08 by EdenZ-2706