FRC2706/2020-2706-Robot-Code #12: Operator Control Sensitivity

nicojopr, 2020-01-11:
complaint is that Mergonaut controls were too sensitive 
what can we do to adjust this? (square inputs etc)
do different controller options help?
can we make this user adjustable?
will need feedback from a 2019 driver.

kevlam2706, 2020-01-13:
This is related to the issue we are investigating for the prototype shooter, which is that drivers want to be able to precisely aim which direction the robot is pointing without overshooting (which can be described as twitchiness).

Ideas: 
- dampen x-axis output dependent on Y-axis value (do not allow large swings in turning when moving faster)?
- ...

kevlam2706, 2020-01-15:
Use Mergonaut for this investigation and get Erik's feedback as he was the driver and is familiar with how the robot handled last year.

If we can find something better that he likes, we can use this information to make the 2020 robot drive better.

nicojopr, 2020-02-08:
Initial version based on software team's drive preferences to be delivered.
New task will be created for any changes identified later by drive team.