FRC2706/2020-2706-Robot-Code #13: Current Limiting

nicojopr, 2020-01-11:
-take an existing robot and apply current limiting profiles to the drivetrain
-goal is to shut down the motors if they're stalling for too long (pushing against opponent) (or apply motor brake?)
-but do not prevent ability to fight back minor skirmishes or recover from crashing into a field obstacle
-add a warning system before motors cut off so that drive team does not get surprised by a sudden power loss?

kevlam2706, 2020-01-13:
Use Mergonaut for this task as we already know how it behaves without the current limiting profiles.

Citysurvivor, 2020-01-15:
Will do after "IR distance sensor", open to collab with other students if anyone wants to.

kevlam2706, 2020-01-15:
Suggest the following resources to help:

https://phoenix-documentation.readthedocs.io/en/latest/ch13_MC.html

Feel free to Google and search places like the Chief Delphi forums to see what other teams are doing for drivetrain current limiting, though I recognize that this is like trying to drink from a firehose.

closed 2020-03-06 by LaraLim