FRC2706/2020-2706-Robot-Code #19: Velocity PIDs for Shooter

nicojopr, 2020-01-11:


nicojopr, 2020-01-11:
SHOOTER TEST

build high goal target
connect shooter to a robot
can we find a speed setting, angle, etc. to hit the target?
is it consistent?

SHOOTER PID

connect shooter to a robot
write software to hold velocity given PID constants from controls team
does this result in more consistent shooting?

nicojopr, 2020-01-11:
Awaiting shooter subsystem

kevlam2706, 2020-01-13:
There is a prototype shooter that is mounted on protobot. I would suggest that this unblocks the task and that we start on this ASAP so we can derisk the concept of maintaining a velocity PID and also characterize things like the recovery time, which will impact how quickly we can rapid-fire 5 balls.

We will need to coordinate with controls to get access to protobot for testing.

kevlam2706, 2020-02-03:
This was demo'd on Saturday using PID values obtained from Controls. @LaraLim can we consider this done?

LaraLim, 2020-02-03:
It has been tested and gives a decent foundation for the actual subsystem, so I would consider it done.

closed 2020-02-03 by LaraLim