FRC2706/2020-2706-Robot-Code #2: Colour Sensor

nicojopr, 2020-01-11:
Using FRC provided colour sensor, write software to access the sensor

nicojopr, 2020-01-11:
connect the color sensor to the roborio. 
write some code to read the colors and print out what it sees.
how easy is it to distinguish different colors?
what distances does this work at?
can we read through clear plexiglass?
how quickly can we react to the color changes (e.g. reading a spinning wheel)
could this be used to read the floor tape colors?

closed 2020-01-11 by nrgill28

georgetzavelas, 2020-01-11:
@nrgill28 Have we tested this and know it works? These tasks are more broad than just merging the code.

nrgill28, 2020-01-11:
I tested the code on the color strip provided in the KOP and it correctly identified each without issue. We'll have to do more testing to answer all the questions above, but as for the subsystem code I think everything is there (Since it's basically just a proxy class for RevRobotics' library)

georgetzavelas, 2020-01-11:
Awesome! Love the additional context.

PR #16 (2020-01-11): Merge ColorSensorSubsystem into master
Added a subsystem for interacting with the color sensor and color matching classes.

Closes #2 .