FRC2706/2020-2706-Robot-Code #20: Operator Control Shoot High

nicojopr, 2020-01-11:


nicojopr, 2020-01-11:
SHOOTER AIM - may or may not also be part of Shoot Low

we will likely need to aim the robot to aim the shooter
write some code to pivot the robot in place based on some input (ultimately will come from vision)
use Pigeon IMU
try to be accurate at small adjustments (up to +/- 20 degrees)

nicojopr, 2020-01-11:
Awaiting shooter subsystem

kevlam2706, 2020-01-13:
I would suggest that this task is really about the aiming of the robot since we already have a task for managing the shooter (PID velocity holding etc). 

Abhi is investigating options to help aim the robot, which could include:

- PID gyro turn at small increments (e.g. turn left 3 degrees)
- joystick axis gets scaled to a small interval of values, just above deadband. e.g. scale from (0,+1.0) to (0.3, 0.5) or whatever values just get the drivetrain moving.
- impart a small forward velocity while doing a slow curve to the left or right, to help overcome stiction.
- instead of arcadeDrive, use tankDrive and leave one side stopped while only turning the other side (swing around an arc rather than spin on the spot).
- ...

georgetzavelas, 2020-01-13:
Is there any vision assist that the vision subteam is exploring as part of the aiming?

kevlam2706, 2020-01-13:
Yes - the hope is that vision will give us the exact direction that we need to turn to face the target.

So if vision says “the target is 4 degrees to the right” we need the robot to be able to turn exactly 4 degrees as quickly as possible.

I actually suspect that it will be easier to turn larger angles than smaller, since larger angles give the PID subsystem more time to react and anticipate. With a small angle if you’ve gone too far already then it’s too late, and you have to turn around and go back - that looks like oscillation.

weizhang-2706, 2020-01-30:
I am also thinking about automatic calculation for the shooter motor's speed. 

After the robot gets aligned to the target, if we know the following values:
1) the shooter's angle of elevation (I assume this is fixed and we can measure)
2) distance between the robot location and the base point of the target ( do we have a distance sensor or can the vision team tell us?)
3) Height of the target (game manual has this value)

Then we can calculate the initial velocity of the power cell, which should be mapped to the shooter motor's speed.
 
The formula will be attached soon.

davidliu-2706, 2020-01-31:
[image]
Formula to calculate the initial velocity when given d, h and theta.

robertlucyshyn, 2020-01-31:
Just to let you know that the Vision Team is working on an approach to calculate the location in (x,y,z) coordinate of the center of the outer target from the so-called "shooter" camera (i.e. the camera mounted near the shooter). From this location, you should be able to calculate whatever you need. You can calculate, for example, the distance from the camera to the center of the outer target, and you can also calculate the "yaw angle" from the line of the sight of the camera to the center of the outer target. Note that you may also want to know this second quantity is order to get the robot face the center of the target. Also note that the quantities we give you are with respect to the camera. If the camera is positioned close to the shooter, that might be good enough---otherwise, you might have to computer some small adjustments to what we give you. Come to see the Vision Team if you need to discuss the exact form of the data you might want from us.

weizhang-2706, 2020-02-01:
This is linked to issue #45 (Shooter subsystems).

kevlam2706, 2020-02-02:
We could try this with the prototype shooter by setting a fixed distance and seeing if the calculated speed lets us hit the target (at known height). We need to understand if the formula will work even with the physical losses of the shooter (friction, air resistance etc).

Kevin

weizhang-2706, 2020-02-03:
That is a good idea, Kevin! We can definitely test using the prototype shooter with a few fixed locations of the robot. 
Since the power cell is light, we could have some loss due to friction and air resistance. I am thinking we could adjust the "golden" point of the target. We will figure it out during the test.