FRC2706/2020-2706-Robot-Code #21: Game Over Signal

nicojopr, 2020-01-11:
aka End Game Signal
controls team would like to know if there is a way we can signal the Pi's to
start shutdown procedure at the end of a match
does FMS send robot code an indicator that the match is over? automatically send something at 5 seconds left? Driver control?
can we send a signal out (i.e. on networktables) once this signal is received?
prefer "end match" over "robot disable" or else we'll have the Pis coming up and down all the time as we practice

davidliu-2706, 2020-01-11:
To do: 
1. Figure out if FMS provides an End Game signal to the driver station software.
    Link to issue #22.
2. If there is a signal, pass on this signal from the driver station software to the robot.
3. Our robot software needs to pass this signal to the control team (raspberry pi) via an ethernet interface.
   Link to issue #27.

kevlam2706, 2020-01-13:
It was suggested that this might already be an FMS-provided networktables value. If so, please find out what it's called, so we can tell the Controls team that they could monitor for this signal themselves.

If it is not, then we will need code in our robot code to propagate this out to them.

weizhang-2706, 2020-01-17:
Yes, NetworkTables are definitely ones which can do this job. 
There is a default NetworkTables instance created automatically when the program started. It provides shared data between FMS/driver station/dashboard and robot. 
Need to investigate:
1) if FMS information can be directly sent to robot or need to be relayed by driver station. 
2) may need create a client-side program running on raspberry pi for control team to communicate with robot. (note: robot is always server side)
3) may need to create a client side program running raspberry pi for vision team to communicate with robot.
4) how to test FMS messages.

weizhang-2706, 2020-01-17:
suggest to create a new one labeled as platform for NetworkTables or Communication between robot and driver station/co-processors.

kevlam2706, 2020-01-24:
Here are some other thoughts on how to accomplish this signal.

https://www.chiefdelphi.com/t/match-ended-software-legality/166352/14

davidliu-2706, 2020-01-25:
- The most efficient method for the end game signal seems to be using the drive station isEnabled/isDisabled status.

- We should also detect if we are in the teleoperated stage, as there could be a few seconds time in between the transition from autonomous to teleoperated.

- We should only control this during a real match.

- We could use the network tables to send the shutdown signals to the raspberry pi.

kevlam2706, 2020-01-26:
Sounds great. Can you write a prototype of what the logic should be? For now you can pretend to send the shutdown signal and just write something to the console or log instead. You can test by using practice mode on the driver station.

Kevin

weizhang-2706, 2020-02-01:
We also need to provide a game end shut down signal and a start game signal for the vision team.
Network table entries: 

- VisionControl/ShutDown

- VisionControl/StartUp

kevlam2706, 2020-02-03:
I think we were also asked to send a Game Start signal.

Please incorporate your logic for both of these into the 2020 codebase. If we're not ready to publish to networktables, at least do a System.out.println() to indicate that this is the point where we would send a Game Start and Game Over signal.

weizhang-2706, 2020-02-03:
Unit test was done. We used networkTables for control and vision. All the network tables can be published to shuffleboard by default.