FRC2706/2020-2706-Robot-Code #22: Feedback to Driver

nicojopr, 2020-01-11:
Investigate ways to provide feedback to the driver via robot software, whether to controller or to bling via network tables, etc..

kevlam2706, 2020-01-15:
Prototyping actions include:
- haptic feedback from controllers
- shuffleboard widgets
- camera viewers
- other stuff?

Get to know how these work and what they're capable of, so that we can use this information to develop the driver feedback for the 2020 robot.

jwt2706, 2020-01-15:
Understood. I was trying to redo what I did on your computer Monday and I
got everything to work except the camera...still don’t know what went wrong
but I’ll try to figure that out today if I have time. Also, I found some
interesting code samples on GitHub that might make the controller rumble, I
haven’t tried it out yet thought.

Citysurvivor, 2020-01-18:
I think it would be a good idea to make sure that the feedback messages are consistent, and to ask the drive team about their preferences - for example, only make the controller rumble when there's an urgent warning from the bot.

Citysurvivor, 2020-02-11:
@jwt2706 Hey Jacob, just a heads up: I'm currently working on prototyping some code for a current limiter, and I would like to give it driver feedback functionality using whatever driver feedback system you come up with, when you're ready. Thanks, Charlie.

(Current limiter code would prevent the robot from burning out its motors on field - I plan on making the code warn the drive team when the robot current limiting occurs).

jwt2706, 2020-02-11:
I’m still waiting to know who will be on the drive team to do any changes,
but if you want we can try things out Wednesday.
We can hook up your code to Shuffleboard or a rumble feature if you like,
just to test it out.

kevlam2706, 2020-02-11:
Jacob, what we are trying to do this year is based on the concept of “agile software development”. In agile, we don’t wait until we know everything before we start making things work. We can get smaller chunks of things working now, then come back and refine more later.

Don’t wait for the drive team. Go ahead and make it work for whoever needs it at the time, which right now is Charlie.

Citysurvivor, 2020-02-12:
I don't need it right away (my code isn't ready yet) but I will need it in the near future once I finish (likely Saturday).

jwt2706, 2020-02-15:
Ok, just let me know when you are ready :)

nicojopr, 2020-03-07:
Intial submission is done, new tweaks can be made via new tasks or as needed without Kanban.  Tying to the Shuffleboard PR