FRC2706/2020-2706-Robot-Code #27: Bling

nicojopr, 2020-01-11:
Controlling colours for:
- fun
- driver feedback

Involves writing to network tables from roboRio to communicate with raspberry Pi

weizhang-2706, 2020-01-17:
After communication between robot and raspberry-pi (control team) is implemented (very possible by NetworkTables), we need to discuss messages sent to control system. In other words, what bling patterns we are going to map to different messages.

kevlam2706, 2020-02-10:
The original concept of the Bling subsystem was to be similar to the 2019 code in which we would have an instance that would communicate over networktables to the external bling system, running on a Pi on the robot.

This year, wpilib added the ability to directly control LED strips from the RoboRio using a PWM output. This may be the final solution for us as it saves complexity and problem-solving on the Raspberry Pi side. However, it means the robot code is fully responsible for animating the LEDs.

If we go with this solution, we will need software. Let's pivot this issue to meet this need. If we go the old way, we already have examples of working code from 2018-2019 we could just bring back in.

Controls will install Bling strips on Mergio. They also exist on Protobot.

For this issue, write the basics of a subsystem that can initialize a set of LED strips (how many LEDs on the strip? This should be configurable as a subsystem constant). Set up a method that can do something simple like change all the LEDs to display a certain color. See this page for details and examples:

https://docs.wpilib.org/en/latest/docs/software/actuators/addressable-leds.html

Don't worry about making a ton of fancy patterns for this first cut of the software. More patterns will be added in a future issue.

weizhang-2706, 2020-02-10:
Agreed. Directly controlling LED strips is more straightforward and can avoid the integration need with Control system. We can first prototype this new way.

kevlam2706, 2020-02-10:
The future issue is #81 😀