FRC2706/2020-2706-Robot-Code #28: Vision System

nicojopr, 2020-01-11:
Develop robot software to interact with vision system, likely via network tables.
This task can be broken down into smaller tasks when known.

kevlam2706, 2020-01-15:
Develop a Vision subsystem, following our robot Subsystem class conventions.

Instead of traditional subsystems which have references to motor controllers and sensors, and commands to operate such, this subsystem will interface to the external vision system, presumably via networktables.

They may want us to feed them robot data such as IMU heading and robot tilt, etc.

We will probably get a feed of robot data from them such as target(s) acquired, distance to target, angle to target, etc.

This information can be used by autonomous and teleop commands such as some kind of aim-and-shoot.

nicojopr, 2020-02-22:
Vision support does not require it's own subsystem but is integrated into the various subsystems it is supporting

closed 2020-02-22 by nicojopr