FRC2706/2020-2706-Robot-Code #5: Rotary Selector Switch

nicojopr, 2020-01-11:


nicojopr, 2020-01-11:
To toggle between Autonomous and Teleop

kevlam2706, 2020-01-20:
The code for the selector switch seems to be buggy after you rotate the switch to 7 or higher. Inspection of the code seems to be fine. The switch itself is rather low-tech so I can't imagine why it would suddenly be failing after several years of successful use, but it's possible. They're analog and low-tech. Could also be a change in how the RoboRio measures and interprets analog input values.

We have access to several analog selector switches - two are on Mergio, I believe one is on Protobot, one is on Mergonaut, and I bought a brand new one which I can bring in. The switches are all numbered and dated by Monique so we know how to tell them apart and how old they are.

I suggest we sit down and take some readings to make sure the switches are still behaving in a consistent way, and also to make sure the range values in our code still match what we actually get.

Here's what I suggest:

1. Start a spreadsheet so we can track readings from multiple switches (columns) at each of their 12 settings (rows).

2. Write code that continuously reads the analog voltage and dumps it to a screen or a log. A System.out.println() would let us see how the values fluctuate over time because you can see the changes as they scroll by, but the Shuffleboard display would also work.

3. Plug in a switch and set it to position 1. Watch the readings on the screen. They'll fluctuate, but they should all be within some value range like 4.10241, 4.10446, 4.09243, 4.13852, etc. 

4. Write down the average value that seems to be measured at this position. For the above example, it looks like it would be about 4.1.

5. Repeat this with all the other positions.

6. Repeat this with other switches.

6. Come up with min/max range values that would comfortably "bracket" each of the averages, for each of the switches. For example, if the average reading seems to be 4.1, then a range of 4.0 to 4.2 would be safe (as long as it doesn't overlap the reading of the next position).

Hopefully the switches are similar enough that we can generate one set of ranges that work with all of them. However, if it does not, then we need to make sure we customize the 2020 code for whatever physical switches are installed on the 2020 robot.

If we discover any switches that are misbehaving (inconsistent readings, or overlapping values) we need to mark them and either discard them or make sure they don't get used on the robot.

LaraLim, 2020-01-21:
[Analog Selector Voltages.xlsx](https://github.com/FRC2706/2020-2706-Robot-Code/files/4088836/Analog.Selector.Voltages.xlsx)

kevlam2706, 2020-02-03:
Closing as this seems to be done.

closed 2020-02-03 by kevlam2706