FRC2706/2020-2706-Robot-Code #31: Constant management on multiple robots

Zinka010, 2020-01-13:
We need some sort of way to not have to change the code every time we deploy on different robots.

A file like robot-config on the roboRIO is one possible solution which worked extremely well in the past.

kevlam2706, 2020-01-13:
I liked how we did it in the past. My only concern is the additional levels of indirection required - if you know what I mean. However, it worked, so feel free to bring it forward as-is unless you prefer a different way.

georgetzavelas, 2020-01-13:
How about exploring the use of a .properties file?

Zinka010, 2020-01-14:
@kevlam2706 I'm not super sure what you mean by indirection. Is it the annoyance of having to SSH into the roboRIO and create the file or more over the having the robot specific generic?

kevlam2706, 2020-01-20:
I take it back. I was thinking of the way we used to do constants
before they did the FluidConstants. There used to be a few arrays that
you had to define your constant (and the keys) before you could
finally use it.

I like what they were trying to do with FluidConstant where these
values were constants, but editable on the Shuffleboard or
SmartDashboard. I don't know if we need ALL of our robot constants to
be this fluid (a fluid constant seems to be an oxymoron) but we should
give some thought to how we can define things like PID values in a way
that we can edit them quickly if we need to do testing and
calibrating.

kevlam2706, 2020-02-03:
This seems to be done and working.

closed 2020-02-03 by kevlam2706