FRC2706/2020-2706-Robot-Code #34: Operator Control: Indexer

nicojopr, 2020-01-18:
The indexer provides power cell storage (4 cells in indexer and one in funnel) and the ability to feed the shooter. It will have one motor with encoder.

nicojopr, 2020-01-30:
@kevlam2706 comment from duplicate :
I just noticed that the Feeder wheel may be separate from the Indexer wheel. In the prototype the two were tied together by a belt. I’ve asked on Slack to clarify if the real one will be separately controlled, in which case we will need to add a second motor to this subsystem and include in the commands the logic to manage the timing and sequencing between them.

kevlam2706, 2020-02-03:
Some terminology update since I keep getting this wrong...

The big wheel inside the hopper is the **Feeder**. Its job is to Feed the shooter.

The little wheel at the edge of the funnel is the **Indexer**. Its job is to ensure the balls enter the hopper with a small separation between each ball so they don't jam up.

The Feeder and Indexer wheels will both be driven by the same motor on a single belt. The Indexer wheel is designed to rotate slower than the Feeder wheel so that each ball ends up spaced out a bit when inside the hopper.

There will be an IR sensor to detect when a ball is sitting at the Indexer. This should trigger the Feeder to "increment" and move everything closer to the shooter by a predetermined distance.

We should be able to keep track of how many balls are in the hopper. This will be used when shooting and also when intaking. If we already have four balls in the hopper and one at the indexer, for example, then we may want to disable the intake so that we don't take any more.

AbhiJ2706, 2020-02-03:
@kevlam2706 I believe we will need another IR sensor at the shooter end of the indexer track in order to monitor the amount of balls in the indexer. This will count the number of outgoing balls, so that that number can be subtracted from the count the first IR sensor creates (which would add 1 for each incoming ball). That way, the exact number of balls in the indexer will known.

kevlam2706, 2020-02-04:
We should check with controls and D&F to see if they are planning on this.