FRC2706/2020-2706-Robot-Code #36: Documentation: Update Software Spec (Driver Feedback)

Citysurvivor, 2020-01-18:
The software spec document should be updated:

Driver feedback should be standardized so that all warnings/notifications make sense to the drive team's preferences. For example, flashing orange lights on the corners = caution, controller vibration = urgent warning (i.e. "motor current cutoff imminent").

georgetzavelas, 2020-01-18:
This is good feedback but to make a change to the spec we need to specify exactly what it should be. Having it being interpreted to drive team's preferences is too subjective to meet as a requirement IMHO. We can of course have a requirement that we can allow driver feedback (we need to specify what that is) to be customizable 😄

nicojopr, 2020-02-06:
For the general design documentation task:
For example of previous software spec, see  last year's  write ups https://drive.google.com/drive/folders/1t3DF-qLRp0zoJyNotzSUO6AjYBazRQiq?usp=sharing

We propose to have each person update the wiki this year for their area of contribution, then we can pull info together. In the meantime, good to start a skeleton doc for 2020 Software in https://drive.google.com/drive/folders/1IvX7M4hOTm0dBTsD9Jj7GpSdslO8-dWY

Citysurvivor, 2020-02-06:
> This is good feedback but to make a change to the spec we need to specify exactly what it should be. Having it being interpreted to drive team's preferences is too subjective to meet as a [...]

True, although now that I reread my issue, I realize that I didn't mean to say that. (My issue was worded poorly.)

nicojopr, 2020-02-15:
Link to template to use for each software area documentation https://docs.google.com/document/d/1yA21zNQh31z1THF45gdJ3XXEMdJyRvJdj1Vx6dU6RnM/edit#heading=h.3as4poj