FRC2706/2020-2706-Robot-Code #37: Vision System: Vision Assisted Power Cell Pickup

nicojopr, 2020-01-18:
The vision system will provide data via network tables to tell the robot how to navigate and pickup a power cell.

RachelLucyshyn, 2020-01-18:
In the text below, note that MergeVision refers to the name of the network table.

When the following Network Table value is set to true:

MergeVision/PowerCell (boolean): Set to true when PowerCell vision assist is being requested, false otherwise.

The vision system is going to provide in "MergeVision" the following Network Table values:

MergeVision/YawToPowerCell (int): Angle in degrees between power cell and line of sight of camera. Positive when PowerCell is located to right of the robot.

MergeVision/DistanceToPowerCell (float): Distance in feet from the camera to the Power Cell.

The Vision team needs code that will set PowerCell to true, read YawToPowerCell and DistanceToPowerCell, and then move the robot to the Power Cell according to those values. One approach may be first to rotate the robot to point to the Power Cell and then move the robot in a straight line to the Power Cell. Other approaches can be considered as well.

Further details can be found in the section Vision Detection of Power Cell in the Vision 2020 Design Document located at 
https://docs.google.com/document/d/1jOhvzP2bSJAUKV0i7xB6ORrvqK-BhbGMSkfR7i_-ims/edit?usp=sharing

georgetzavelas, 2020-01-18:
Sounds cool @RachelLucyshyn ! Can you make the doc viewable for everyone 😄

RachelLucyshyn, 2020-01-18:
The link should work now