FRC2706/2020-2706-Robot-Code #48: Indexer Subsystem

nicojopr, 2020-01-25:
Production software for the indexer subsystem in the robot code, including:
- interface definitions, motors, sensors
- commands
- etc..
- tested independently
- merged and integrated with 2020 robot code in github
- ready for integration

kevlam2706, 2020-01-26:
I just noticed that the Feeder wheel may be separate from the Indexer wheel. In the prototype the two were tied together by a belt. I’ve asked on Slack to clarify if the real one will be separately controlled, in which case we will need to add a second motor to this subsystem and include in the commands the logic to manage the timing and sequencing between them.

nrgill28, 2020-01-29:
Is this a dupe of #34

nicojopr, 2020-01-30:
Duplicate of #34 ...closing

closed 2020-01-30 by nicojopr