FRC2706/2020-2706-Robot-Code #51: List of Autonomous Modes

nicojopr, 2020-01-30:
Brainstorm with previous drive team and other interested members on potential mode of operations desired for autonomous mode (first 15 seconds of the match, no driver). Scenarios that have been mentionned at various team meetings so far:

- basic drive off the initiation line
- score 3 high power cells (pre-loaded)
...

Outcome of this task is a prioritized list of modes that could be made available for competition

nicojopr, 2020-02-04:
Add the list to our wiki and request review/discussion at a software team meeting (Wed?)

Zinka010, 2020-02-06:
So another obvious one is do nothing and wait then high goal

Zinka010, 2020-02-06:
List avaialable at https://github.com/FRC2706/2020-2706-Robot-Code/wiki/Autonomous-Modes

kevlam2706, 2020-02-10:
I made some edits to the list to add "move off initiation line" to the high goal autos. We probably want those 5 points :smiley:

nicojopr, 2020-02-11:
This documentation task is considered done, new tasks are created for the specific autonomous modes

closed 2020-03-07 by nicojopr