FRC2706/2020-2706-Robot-Code #59: Vision System: Vision Assisted Shooter

nicojopr, 2020-02-04:
Specific software for vision assisted shooter aim, spinup

Production software for the vision subsystem in the robot code, including:
- interface definitions (i.e. network tables)
- commands
- tested independently
- merged and integrated with 2020 robot code in github
- ready for integration
etc..

davidliu-2706, 2020-02-15:
[image]
When we are given the distance from the target, the height of the target and the angle at which the ball leaves the robot, we can calculate the velocity of the ball.

davidliu-2706, 2020-02-15:
[image]
This formula can be used to calculate the revolutions per minute of the shooter wheel when we are given its velocity.