FRC2706/2020-2706-Robot-Code #87: Prototyping: NEO-based arm position control

kevlam2706, 2020-02-12:
We have been asked to "investigate", "_in theory_", what the software (and PID tuning process etc) would look like if the arm was controlled by a Rev Robotics NEO brushless motor instead of a CIM, and driven by a Spark Max instead of a Talon SRX.

Rev Robotics has an example of position-based closed loop control using the NEO (and it's built in encoder), a separate example of using a limit switch, and a separate example of "Smart Motion", here:

https://github.com/REVrobotics/SPARK-MAX-Examples/tree/master/Java

Suggested process for this prototyping task:
1. Get a NEO and Spark Max that have been "connectorized" by Controls and ready to plug into a robot chassis (power and CAsuch as Cosmobot. For this investigation, it would be good to have the NEO attached to a VersaPlanetary gearbox with a relatively high gearing (slower output RPM), either standalone or maybe using the existing winch prototype board.

2. Deploy the basic position control example and see that it works as-is. Some PID tuning may be needed; the examples let you play with values directly in Smart Dashboard, or you may try using the (frickin' awesome) Spark Max Client application on your computer and control the Spark Max directly via USB.

3. Add limit switch functionality (will require some controls work to connect a switch to the Spark Max data port, assuming we have whatever connector interface is needed)

4. Verify that limit switch functionality works to stop travel at the bottom of the range.

5. Verify that we can set and enforce a soft limit at the top of the range.

6. Add some fixed setpoints representing multiple arm positions and verify that we can jump back and forth between them in a smooth and controlled manner.

7. Maybe have a look at the "Smart Motion" example and see if it's better than plain position control.

8. Can we set the encoder values to an arbitrary position at startup? Can we set the limit switch to zero the encoder values? Does that work when the robot is disabled? We still need to plan for the scenario of the drive team starting the arm in the "up" position at the start of the match.

kevlam2706, 2020-02-13:
This is low priority at the moment.

nicojopr, 2020-02-29:
Cancelled, not required for 2020 robot

closed 2020-02-29 by nicojopr